dewet22/givenergy-modbus #11: GivEnergyClient missing from code?

HLFCode, 2022-10-07:
* GivEnergy Modbus version: N/A
* Python version: N/A
* Operating System: N/A

### Description

**Describe what you were trying to get done**.
Fork or download the code

**Tell us what happened, what went wrong, and what you expected to happen.**
The downloaded or forked code is not the same as that installed by pip install.

This is probably my ignorance of github but the code here does not seem complete and is structured differently to the `pip install givenergy-modbus` code
E.g. the main class is GivEnergyClient but I can't find it anywhere in the code...

This is a great project and I have a couple of minor mods I would like to contribute.

Thanks
Mike

springfall2008, 2022-12-25:
I have the same issue?

springfall2008, 2022-12-25:
Using the latest Tag from March works for me

closed 2026-05-11 by dewet22