dewet22/givenergy-modbus #19: Intermittent results when querying the Inverters

lodge-rj99, 2023-01-20:
* GivEnergy Modbus version:  givenergy-modbus-0.10.1
* Python version: 3.11
* Operating System: MacOS 12.6.2

### Description

Describe what you were trying to get done.

Run this simple python script in PyCharm IDE to grab the battery and inverter info. Would just expect a  clean exit

```
import json
import logging
import datetime
import time
from givenergy_modbus.client import GivEnergyClient
from givenergy_modbus.model.inverter import Model
from givenergy_modbus.model.battery import Battery
from givenergy_modbus.model.plant import Plant
from os.path import exists
import os

client = GivEnergyClient(host='192.168.0.70')

p = Plant(number_batteries=1)

client.refresh_plant(p, full_refresh=True)
GEInv = p.inverter
GEBat = p.batteries
```
Tell us what happened, what went wrong, and what you expected to happen.

Sometimes it works and sometimes it doesn't, it's completely random.

When it works it exits cleanly (and if I add the code to print out an inverter or battery setting, it prints the correct value)

When it fails I see this

> Did not receive expected response type: ReadHoldingRegistersResponse != ReadInputRegistersResponse
> Returned base register (0) does not match that from request (60).
> Traceback (most recent call last):
>   File "/Users/richard/PycharmProjects/pythonProject3/venv/giv.py", line 17, in <module>
>     GEInv = p.inverter
>             ^^^^^^^^^^
>   File "/Users/richard/PycharmProjects/pythonProject3/venv/lib/python3.11/site-packages/givenergy_modbus/model/plant.py", line 32, in inverter
>     return Inverter.from_orm(self.inverter_rc)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "pydantic/main.py", line 578, in pydantic.main.BaseModel.from_orm
>   File "pydantic/main.py", line 1056, in pydantic.main.validate_model
>   File "/Users/richard/PycharmProjects/pythonProject3/venv/lib/python3.11/site-packages/givenergy_modbus/model/register_getter.py", line 69, in get
>     return getattr(self._obj, key, default)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/richard/PycharmProjects/pythonProject3/venv/lib/python3.11/site-packages/givenergy_modbus/model/register_cache.py", line 26, in __getattr__
>     val = self[register]
>           ~~~~^^^^^^^^^^
> KeyError: HR:120

The ability to query the inverters and batteries directly in this way is very interesting to us, so I'd love to get this working and stable if I can

Wondering if it could be poor comms quality as our 3 inverters have wifi dongles for connectivity.

britkat1980, 2023-01-23:
The modbus response from the GE invertors is not 100% stable, so it is to be expected that you will occasionally get these failures. The code you write to use this library will need to account for this.
When I use this library in [GivTCP](https://github.com/britkat1980/giv_tcp) I see this behaviour too and handle it by holding the cache and retrying.

lodge-rj99, 2023-01-26:
That’s most helpful. Good to know it’s not an issue unique to us. Many thanks for the information. Regards

dewet22, 2026-05-11:
Intermittent results are typically caused by running multiple concurrent pollers against the inverter — the inverter doesn't handle concurrent Modbus connections well and becomes unstable under the load. If you have more than one consumer (e.g. GivTCP alongside another integration), use `passive=True` in `watch_plant()` for the secondary consumers so that only one client is actively sending requests at a time. Closing as addressed by the passive mode — please reopen if you're seeing this with a single poller.