dewet22/givenergy-modbus #26: Need a fresh release

SamKua, 2023-03-28:
This code is in use in another givenergy integration, which is broken for us due to the missing serial number prefix. Apparently the other integration requires a release on this in order to use the fix which my partner submitted (https://github.com/dewet22/givenergy-modbus/commit/01224c77bab68309e06c0dd7e946dfe41b28de22) which has been merged.

Its been over a year since a new version was released. Could one be done? 
Thanks.

veehexx, 2023-07-06:
we could really do with this @dewet22.
This is used in HomeAssistant so every HA update reverts it back to an unfixed version (i assume v0.10.1) thus each update means manually re-applying the EA prefixed inverter. While simple to do, it is a faff updating it multiple times a month with each HA update. Looks like theres also requirement for the [latest gen3](https://github.com/dewet22/givenergy-modbus/issues/34) too.

the EA gen2 fix has been solid for me for months.

a side point; i updated HA a few days ago and it appears to have dropped the 'cls.' and potentially moved to elsewhere in the code (https://github.com/dewet22/givenergy-modbus/commit/01224c77bab68309e06c0dd7e946dfe41b28de22). not sure whats going on there

britkat1980, 2023-07-06:
> we could really do with this @dewet22. This is used in HomeAssistant so every HA update reverts it back to an unfixed version (i assume v0.10.1) thus each update means manually re-applying the EA prefixed inverter. While simple to do, it is a faff updating it multiple times a month with each HA update. Looks like theres also requirement for the [latest gen3](https://github.com/dewet22/givenergy-modbus/issues/34) too.
> 
> the EA gen2 fix has been solid for me for months.
> 
> a side point; i updated HA a few days ago and it appears to have dropped the 'cls.' and potentially moved to elsewhere in the code ([01224c7](https://github.com/dewet22/givenergy-modbus/commit/01224c77bab68309e06c0dd7e946dfe41b28de22)). not sure whats going on there

Are you using GivTCP? If so it shouldn't use the default version, I provide a forked version of this library which calculates Inverter stats without using the serial number...

veehexx, 2023-07-06:
> Are you using GivTCP? If so it shouldn't use the default version, I provide a forked version of this library which calculates Inverter stats without using the serial number...

I'm using the https://github.com/cdpuk/givenergy-local, which states it uses this version. (cdpuk is waiting for an updated release)
I'm open to moving to GivTCP, but was on the impression givenergy-local was the only solution to have local polling and was the current recommended addon to use.

dewet22, 2023-07-10:
The v0 branch is really unsupported at this point. It isn't a good implementation, and suffers a lot from timing subtleties because of its synchronous implementation. The v1 release is completely async and will play better with multiple clients talking to the inverter. This library has always intended to be a base-line implementation, with clients to be built on top of it. I have a prototype HASS integration working very reliably with it, but it'd be trivial to add on more features (some of which I want) like MQTT and influxdb support.

JRascagneres, 2023-11-22:
@dewet22 Can you provide some example code for the v1 branch? With `GivEnergyClient` not existing anymore