dewet22/givenergy-modbus #28: pymodbus 3.x support

FileGo, 2023-05-03:
### Description

I'm using [givenergy-local](https://github.com/cdpuk/givenergy-local) integration in Home Assistant, which uses this library. I also use [home-assistant-solaredge-modbus](https://github.com/binsentsu/home-assistant-solaredge-modbus) integration to monitor my Solaredge inverter.

The Solaredge integration (and the -multi one as well) uses pymodbus >= 3.1.1. Is there any chance to upgrade this project to use a more modern version of pymodbus (currently `^2.5.3`)?

Thanks.

dewet22, 2023-05-11:
@FileGo you'll be glad to hear that in the next major release currently being worked on, the dependency on pymodbus is [completely removed](https://github.com/dewet22/givenergy-modbus/commit/0d8ec13913f23df3c7b9dfcfcf075d45ae813094).

FileGo, 2023-07-21:
@dewet22 Is there any rough ETA on this?